RishiGupta12/SerialPundit #4: OS X 10.9.5 PortListener and DataListener

artjomg, 2015-06-22:
Hi,

I am on mac OS X 10.9.5 and running into issues with listeners when the USB port is disconnected.
I create and register  the required listeners via, where app implements the ISerialComPortMonitor as following:

		scm = new SerialComManager();
		System.out.println(scm.getOSType());
		System.out.println(scm.getLibraryVersions());
		dataListener = new DataListener();
		eventListener = new EventListener();
		comPortHandle = scm.openComPort("/dev/cu.usbmodem1421", true, true, true);
		scm.configureComPortData(comPortHandle, DATABITS.DB_8, STOPBITS.SB_1, PARITY.P_NONE, BAUDRATE.B57600, 0);
		scm.configureComPortControl(comPortHandle, FLOWCONTROL.NONE, '$', '$', false, false);
		scm.registerPortMonitorListener(comPortHandle, app);	
		scm.registerDataListener(comPortHandle, dataListener);
		scm.registerLineEventListener(comPortHandle, eventListener);

I receive the disconnect/connects events properly, but right after the disconnect ISerialComDataListener starts to flood with errors at:
@Override
	public void onDataListenerError(int arg0) {
		System.out.println("onDataListenerError called");
	}

I tried couple of things to unregister and reregister the listeners (event,data) which just guides to detachment of listeners and registration exceptions. Reconnection to the port fails, since the port is still open. Rerun and create new scm instance helps, but detaches the listeners on and I get "Listener not registered" errors on next USB detachment.

How can I avoid the onDataListenerError calls or am I missing something?

Expected behavior.
1. USB connected
2. Create SCM instance and register all listeners Port,Serial,Data
3. Unplug USB -> Port event == 2 should result in stopping serial and data Listeners from interfering with a dead scm connection
4.Reconnect USB -> Port event == 1 properly should reuse register serial and data listeners without errors.

Another remark: I can kill the JVM by simply trying to unregister the portListeners when onPortMonitorEvent == 2 is called. Which is a stupid circular parent destroy access I know, but it should be handled properly without killing the VM. Again all OSX.

Thanks,
Rtg

RishiGupta12, 2015-06-23:
Please raise (just copy -paste issue) on google group and let us follow it there.

- Version mismatch is not an issue.

- The error reporting from dataListener i.e. method onDataListenerError(int arg0) gets called whenever there is an error. Because USB device is removed error continue to occur. This is why this method is called again and again. The onDataListenerError is intended to trigger system recovery policy to write recoverable and fault tolerant application. Read javadoc here http://www.javadoc.io/doc/com.embeddedunveiled/scm/1.0.3

- SCM does not allow to unregister any listener until port which it is monitoring is closed. But I can see that something more graceful is needed here.

- If the usb device is removed and plugged again into system, it is seen as a new device by both OS and SCM. Device may or may not be assigned same name for example if previously it was ttyUSB0 now it may be ttyUSB1. So if there is a registered data listener it can not automatically open newly plugged device and start watching it.

- No need of creating new instance of SCM or playing with JVM. I am going to resolve the issue more gracefully.

RishiGupta12, 2015-06-23:
This test is passing https://github.com/RishiGupta12/serial-communication-manager/blob/master/tests/test56-dl-removePort/src/test56/Test56.java

Please take a look at when error occurs, 1st unregister listener(s) and then close port. Now if you plugin the device again, open port and register listener.

artjomg, 2015-06-23:
Moved to Google group Discussion
https://groups.google.com/forum/#!topic/serial-communication-manager/Bvi5xfyjix0

closed 2015-06-23 by artjomg