RishiGupta12/SerialPundit #5: Windows DLL depends upon debug DLL

andrewdmay, 2015-07-22:
Based upon this discussion: https://groups.google.com/forum/#!topic/serial-communication-manager/pZIrrdn43nY it appears that the windows DLL depends upon MSVCR120D.DLL (as well as IESHIMS.DLL).

I'm not a Windows C++ developer, but from what I've read it appears that the *D.DLL files are debug only and not redistributable, and I've been unable to figure out how to get a copy of MSVCR120D.DLL (I installed VS 2013 Community edition, but still can't find the DLL anywhere - not sure if I need to build something that uses it?).

Ideally the binaries included with the Java library should not require additional DLLs to be installed - but if this is not possible it would be preferable if the libraries were available from the Microsoft Visual C++ redistributable download.

andrewdmay, 2015-07-22:
I built the windows binaries using Visual Studio Community (2015), and I statically linked everything using the Multi-threaded (/MT) option in Configuration Properties > C/C++ > Code Generation > Runtime Library, and it seems to work.

RishiGupta12, 2015-07-23:
- All the build guides are on wiki http://www.embeddedunveiled.com/
- Install Visual Studio Express 2013 with Update 4
- this project is getting feature rich day by day and therefore I have to use debugging libraries to facilate development. From next release onwards I may start providing production release version.
- keep in touch with me as I am making changes / addition in API. Tell me your application and I will keep you informed if my work is going to affect you.
- Please note that repo contains 1.0.3 and there are significant improvements in 1.0.4. I will give you jar for your OS as official release may take time.

RishiGupta12, 2016-01-13:
From 1.0.4 scm.jar provided now will not depend upon any debug libraries.

closed 2016-01-13 by RishiGupta12