RishiGupta12/SerialPundit #8: Error executing

MaXwEllDeN, 2016-04-29:
I am trying to execute the Example here https://github.com/RishiGupta12/serial-communication-manager, in **Getting Started**.

> Exception in thread "main" java.lang.UnsatisfiedLinkError: /tmp/scm_tuartx1/linux_1.0.4_x86_64.so: libudev.so.0: Não é possivel abrir arquivo de objetos compartilhado: Arquivo ou diretório não encontrado
> 	at java.lang.ClassLoader$NativeLibrary.load(Native Method)
> 	at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1941)
> 	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1824)
> 	at java.lang.Runtime.load0(Runtime.java:809)
> 	at java.lang.System.load(System.java:1086)
> 	at com.embeddedunveiled.serial.internal.SerialComPortJNIBridge.loadNativeLibrary(SerialComPortJNIBridge.java:335)
> 	at com.embeddedunveiled.serial.SerialComManager.<init>(SerialComManager.java:581)
> 	at Main.main(Main.java:11)

RishiGupta12, 2016-04-29:
- Which OS you are using.
- Install libudev as shown below:
$sudo apt-get install libudev-dev

flbauer, 2016-05-22:
Hello Rishi,
I am trying on Windows 7 (32-bit) and get an analogous error:
Exception in thread "main" java.lang.UnsatisfiedLinkError: C:\Users\Florian\AppData\Local\Temp\scm_tuartx1\windows_1.0.4_x86.dll: Can't find dependent libraries
	at java.lang.ClassLoader$NativeLibrary.load(Native Method)
	at java.lang.ClassLoader.loadLibrary0(Unknown Source)
	at java.lang.ClassLoader.loadLibrary(Unknown Source)
	...

Cannot find the dll in the internet? Is there an udev for Windows?
An advice would be welcome.

RishiGupta12, 2016-05-23:
Hi, For windows please use dependency walker to find dependent libraries you need to install. The SCM uses minimal libraries that should be already installed on your system. Also share the output of dependency walker with me so that I can analyze and help you.

https://groups.google.com/forum/#!topic/serial-communication-manager/pZIrrdn43nY

closed 2016-05-31 by RishiGupta12

marccollin, 2017-01-15:
Have you found windows_1.0.4_x86.dll file?