RishiGupta12/SerialPundit #10: XP Compatibility

theDuckIsAlive, 2016-06-06:
Is SCM compatible to Windows XP ?
I tried to run the prebuilt-release scm-1.0.4.jar on Windows XP SP3 but I can't get it work.

	Exception in thread "main" java.lang.UnsatisfiedLinkError: 
	C:\Users\xxx\AppData\Local\Temp\scm_tuartx1\windows_1.0.4_x86.dll: Can't find dependent libraries
        at java.lang.ClassLoader$NativeLibrary.load(Native Method)
	at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1937)
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1822)
	at java.lang.Runtime.load0(Runtime.java:809)
	at java.lang.System.load(System.java:1086)
	at com.embeddedunveiled.serial.internal.SerialComPortJNIBridge.loadNativeLibrary(SerialComPortJNIBridge.java:336)
	at com.embeddedunveiled.serial.SerialComManager.<init>(SerialComManager.java:582)
	at demo.Demo.main(Demo.java:14)

I added the missing MSVCR120.DLL to system32 directory but 
still Dependency walker shows a problem with the following 2 DLLs.

	ADVAPI32.DLL  (RegGetValueW  *function is marked red)
	SETUPAPI.DLL  (SetupDiGetDevicePropertyW *function is marked red)

RishiGupta12, 2016-06-07:
- SCM is neither intended to support nor has been tested with windows xp.

- The SetupDiGetDevicePropertyW API is available after windows vista only https://msdn.microsoft.com/en-us/library/windows/hardware/ff551963(v=vs.85).aspx

Therefore in windows XP it will not work. Why you are using windows XP.

theDuckIsAlive, 2016-06-07:
I still use XP for some electronic projects.
That means I have to stick around with the old RXTX library for a while.

closed 2016-06-07 by theDuckIsAlive