RishiGupta12/SerialPundit #11: Can not get shared library spcomlnxarmv7hf.so resource as stream from sp-tty.jar file

hbjoylee, 2016-06-28:
Hi,
When I run my java application on Raspberry Pi, it show errors below, can you help me compile a new sp-tty.jar for raspberry pi, thanks.

Exception in thread "main" com.serialpundit.core.SerialComException: /tmp/sp_tuartx1/spcomlnxarmv7hf.so
        at com.serialpundit.serial.internal.SerialComPortJNIBridge.loadNativeLibrary(SerialComPortJNIBridge.java:322)
        at com.serialpundit.serial.SerialComManager.<init>(SerialComManager.java:367)
       ...................
Caused by: com.serialpundit.core.SerialComException: Can not get shared library spcomlnxarmv7hf.so resource as stream from sp-tty.jar file !
        at com.serialpundit.serial.internal.SerialComPortJNIBridge.loadNativeLibrary(SerialComPortJNIBridge.java:319)
        ... 3 more

RishiGupta12, 2016-06-28:
Latest jars are here https://github.com/RishiGupta12/SerialPundit/tree/master/prebuilt-release

hbjoylee, 2016-06-28:
Thanks Rishi, it solved my problem.

Patrick-w, 2017-08-11:
Hello @RishiGupta12 , 

I'm getting the exact same error, I'm currently running my java app on Ubuntu.
com.serialpundit.core.SerialComException: No such file or directory
        at com.serialpundit.serial.internal.SerialComPortJNIBridge.openComPort(Native Method)
        at com.serialpundit.serial.SerialComManager.openComPort(SerialComManager.java:618)
        at com.wavemark.pythontest.serial_port.main(serial_port.java:15)
I already updated the jars in vain. 

Any idea why? Thank you!

RishiGupta12, 2017-08-11:
The serial port device node does not exist: No such file or directory

closed 2017-08-14 by RishiGupta12