RishiGupta12/SerialPundit #12: Could not identify CPU architecture

hbjoylee, 2016-06-28:
Hi Rishi,

I am using your latest sp-core.jar and sp-tty.jar, but after I run my application it show error below,
Could you please give me some advice that I can identify the issue, thanks.

Exception in thread "main" com.serialpundit.core.SerialComException: Could not identify CPU architecture. Please report your environemnt to us so that we can add support for it !
        at com.serialpundit.serial.SerialComManager.<init>(SerialComManager.java:353)

Also here is part content in my raspberry pi 3's /proc/cpuinfo. 

processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

RishiGupta12, 2016-06-28:
Can you please send me output of System.out.println("OS : " + " System.getProperty("os.arch"));
on your pi3.

RishiGupta12, 2016-06-28:
Please test updated sp-core.jar https://github.com/RishiGupta12/SerialPundit/tree/master/prebuilt-release

hbjoylee, 2016-06-28:
Hi Rishi,
my pi 3 output is OS : arm

hbjoylee, 2016-06-28:
Hi Rishi,

I think I found the root cause.
getCPUArch() in SerialComPlatform.java  should check each line in /proc/cpuinfo, but in fact it only check the first line then return SerialComPlatform.ARCH_UNKNOWN.

I add some prints in SerialComSystemProperty.getOSName() and SerialComPlatform.getCPUArch(), the output shows below:
OS name is: linux
Is Linux
os arch is:arm
Property is processor   : 0

Exception in thread "main" com.serialpundit.core.SerialComException: Could not identify CPU architecture. Please report your environemnt to us so that we can add support for it !
        at com.serialpundit.serial.SerialComManager.<init>(SerialComManager.java:353)

closed 2017-02-14 by RishiGupta12