RishiGupta12/SerialPundit #15: Shared library not in sp-tty.jar

FlaegelEmbed, 2017-01-05:
Hi!

I want to test my USB-Relay/CP2104 with your test77 example. Unfortunately the lib cannot load the shared lib:

"Can not get shared library spcp210xrwinx64.dll resource as stream from sp-tty.jar file !"

Indeed, in the sp-tty.jar there is no such file. Currently not implemented?

FlaegelEmbed, 2017-01-05:
Thanks for response!

it's a finished USB-relay (Conrad Components 393905). I thought it will use uart to control the gpio. But you have to use the manufacturer dll under windows/JAVA ..
So the lib cannot handle the gpios of a CP2104 with the CP210xRuntime.dll?

RishiGupta12, 2017-01-05:
- It is possible to handle gpio of cp2104 using runtime dll
- I will prepare and send new jar to you tonight

FlaegelEmbed, 2017-01-05:
This sounds great, thanks :).
Cant't wait to test it.

RishiGupta12, 2017-01-05:
- Please use the files attached. If any issue occurs please raise it on google group https://groups.google.com/forum/#!forum/serialpundit

[sp-jar.zip](https://github.com/RishiGupta12/SerialPundit/files/687653/sp-jar.zip)

- I have also updated test 77

closed 2017-01-05 by RishiGupta12