RishiGupta12/SerialPundit #17: JNI call ThrowNew failed to throw exception

KanameMashiro, 2017-02-10:
Hello Rishi,
I tried to run the example but I got this error in the Eclipse console, what does it mean or why it happens?

```
JNI call ThrowNew failed to throw exception !, FormatMessageA(), error code : 15100
com.serialpundit.core.SerialComException: 
	at com.serialpundit.serial.internal.SerialComPortJNIBridge.openComPort(Native Method)
	at com.serialpundit.serial.SerialComManager.openComPort(SerialComManager.java:618)
	at paquetePruebaSerial.Example.main(Example.java:14)

The line 14:
            long handle = scm.openComPort("COM5", true, true, true);

I was working on a machine with Win8.1 (AMD64)

RishiGupta12, 2017-02-10:
Does com5 exists. Which serial hw you are using.

KanameMashiro, 2017-02-13:
Thanks for answer, COM5 exists. I'm using an Arduino Nano CH340G.
Solved already, re-installed drivers, that seems to be buggy, and the program works.

closed 2017-02-14 by RishiGupta12