RishiGupta12/SerialPundit #20: Application Crash when starting multiple instance even using new prebuilt releases

combat007, 2017-04-20:
Hi 
Thanks for maintaining this great library.
I Imported the application serial terminal emulator in the application directory in my eclipse workspace,
and opened two instance. As soon as I clicked open button selecting one of the port the other instance terminated, throwing a fatal error and producing an error file. I tried it on Windows as well as on Linux both 32bit and 64bit platform and the problem is same. I am attaching the eclipse console output and the generated error file. Kindly look into the matter.

Eclipse Console Output:-

#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00003986, pid=8279, tid=1742400320
#
# JRE version: OpenJDK Runtime Environment (7.0_91-b02) (build 1.7.0_91-b02)
# Java VM: OpenJDK Server VM (24.91-b01 mixed mode linux-x86 )
# Derivative: IcedTea 2.6.3
# Distribution: Ubuntu 14.04 LTS, package 7u91-2.6.3-0ubuntu0.14.04.1
# Problematic frame:
# C  0x00003986
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /workspace/Serial terminal emulator/hs_err_pid8279.log
#
# If you would like to submit a bug report, please include
# instructions on how to reproduce the bug and visit:
#   http://icedtea.classpath.org/bugzilla
#

[hs_err_pid8279.log.zip](https://github.com/RishiGupta12/SerialPundit/files/941653/hs_err_pid8279.log.zip)

RishiGupta12, 2017-04-23:
I was not in India, just came back. I will look and get back to you. Were you running BOth the instances from within eclipse or how two instances were run.

combat007, 2017-04-23:
Hi nice to see you back.

The crash thing is same if I run it via eclipse or I make it a runnable jar from eclipse and run it on the system platform using the command "java -jar <serial terminal emulator>" on Linux or "Open with Java Runtime" on Windows. Whenever I open two instance as soon as I Open the connection the first instance termintes.

RishiGupta12, 2017-04-30:
Hi, 
The demo app uses default constructor scm = new SerialComManager(); in file SerialTerminalApp.java
When two instances are run 2nd override the shared libraries saved by 1st instance on harddisk therefore crash will occur.
Please use another constructor : SerialComManager(String loadedLibName, String directoryPath, final boolean createDirectory, boolean hotDeploy) specifying 1st 3 arguments as needed.

combat007, 2017-04-30:
Great

It worked like a charm...

closed 2017-04-30 by combat007