RishiGupta12/SerialPundit #21: Add support for armv7 to sp-usb.jar

akuhtz, 2017-04-23:
Hi Rishi,

Could you add support for ARMv7 to sp-usb.jar? Running on a Bananapi it complains about missing spusblnxarmv7hf.so . 

Regards,
Andreas

akuhtz, 2017-04-29:
Hi Rishi,

The same error happens on a Raspberry 3. 
BTW: Open the serial port works fine.

Remboooo, 2017-05-23:
I'm running into the same issue. When I figured I would compile the native part for this platform myself, I could not find any native sources in this repo. Is that intentional? The JNI build scripts just produce the headers, nothing else.

RishiGupta12, 2017-05-31:
May you please try new sp-usb.jar file https://github.com/RishiGupta12/SerialPundit/tree/master/prebuilt-release
I have added armv7hf shared library file.

akuhtz, 2017-06-06:
Works fine on a Raspberry 3 now. Thank you 🥇

closed 2017-06-06 by akuhtz