RishiGupta12/SerialPundit #24: Cannot make it work on Windows 10 x64

AndreiPiakhota, 2017-07-05:
Hi
The library works without a hitch on Linux but I cannot make it work on Windows 10 AMD x64. Every time I try ranning it, the JVM crashes with EXCEPTION_ACCESS_VIOLATION (0xc0000005). I tried on 3 different machines with the same crash. I checked all the libraries (msvcr120.dll, setupapi.dll, advapi32.dll, kernel32.dll) and there they are. There is no consistency in crash log files (hs_err_pid...log). I attach some of these files. It seems like Windows 10 (x64) is not supported but how can I be sure. 

[hs_err_pid5820.txt](https://github.com/RishiGupta12/SerialPundit/files/1125235/hs_err_pid5820.txt)
[hs_err_pid8624.txt](https://github.com/RishiGupta12/SerialPundit/files/1125234/hs_err_pid8624.txt)
[hs_err_pid8996.txt](https://github.com/RishiGupta12/SerialPundit/files/1125236/hs_err_pid8996.txt)

RishiGupta12, 2017-07-06:
Is the issue is with API writeBytesDirect only or you are not able to run anything at all. Were simple read and write methods working.

AndreiPiakhota, 2017-07-06:
Yes, simple write and read work. If I try reading stream of bytes that is call readBytes repeatedly in cycle, I also have problems whereas similar operation works on Linux. So, the problem mostly with methods writeBytesDirect and readBytesDirect in SerialComManager. Thanks for your prompt reply.

RishiGupta12, 2017-07-15:
Is this test working for you https://github.com/RishiGupta12/SerialPundit/tree/master/tests/test61-directwrite/src/test61

RishiGupta12, 2017-08-14:
Hi AndreiPiakhota, 
Is this test working for you https://github.com/RishiGupta12/SerialPundit/tree/master/tests/test61-directwrite/src/test61