RishiGupta12/SerialPundit #25: com.serialpundit.core.SerialComException: Can not get shared library spcomlnxarmv6hf.so resource as stream from sp-tty.jar file !

Abu-Abdullah, 2017-07-06:
i got this exception with the lates build.
```com.serialpundit.core.SerialComException: Can not get shared library spcomlnxarmv6hf.so resource as stream from sp-tty.jar file !```

spcomlnxarmv6hf.so is not there in the jar file

Abu-Abdullah, 2017-07-07:
please add support for ARMv6, it seems raspberry pi zero is only ARMv6

RishiGupta12, 2017-07-07:
It should be armv7. Anyway which toolchain you are using any link.

Abu-Abdullah, 2017-07-07:
can you explain more what i need to do about here (toolchain ?), i'm just using the standard raspbian JESSIE latest build.

 Raspberry Pi zero W is armv6:
https://en.wikipedia.org/wiki/Raspberry_Pi

Abu-Abdullah, 2017-07-11:
compiler is the 4.9

```
pi@raspberrypi:~ $ arm-linux-gnueabihf-gcc --version
arm-linux-gnueabihf-gcc (Raspbian 4.9.2-10) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.28+ #998 Mon May 15 16:50:35 BST 2017 armv6l GNU/Linux
```

Abu-Abdullah, 2017-07-29:
Will it be planned, the support for armv6 ?

RishiGupta12, 2017-07-30:
Try new sp-tty.jar  https://github.com/RishiGupta12/SerialPundit/tree/master/prebuilt-release
I do not have rpi0, therefore you have to try and update me whether it works or not.

What is your project.

Abu-Abdullah, 2017-07-30:
many thanks, i can see that the error is gone. my app is still facing few issues but i believe it is related to configuring the UART in rpi0. i will open another case if i faced issue with the serial library.

my app is controlling flight controller (quadcopter) from serial via raspberry.

again many thanks for this update.

closed 2017-07-30 by Abu-Abdullah