RishiGupta12/SerialPundit #26: SerialComManager.setRTS() causes exception

akuhtz, 2017-07-27:
Hi Rishi,

I'm running Test23 under Win7 64bit and it causes an exception when setRTS() is called (scm.setRTS(DTE, true); )

```
com.serialpundit.core.SerialComException: The data area passed to a system call is too small.

	at com.serialpundit.serial.internal.SerialComPortJNIBridge.setRTS(Native Method)
	at com.serialpundit.serial.SerialComManager.setRTS(SerialComManager.java:1505)
	at test23.Test23.main(Test23.java:65)
ERROR: JDWP Unable to get JNI 1.2 environment, jvm->GetEnv() return code = -2
JDWP exit error AGENT_ERROR_NO_JNI_ENV(183):  [util.c:840]

```

RishiGupta12, 2017-07-30:
Looks like it is JVM side thing not serialpundit. Are you debugging something using a debugger.

http://bugs.java.com/view_bug.do?bug_id=6476706
https://stackoverflow.com/questions/27500398/debugging-java-in-eclipse-error
https://www.eclipse.org/forums/index.php/t/1069715/

akuhtz, 2017-07-31:
Hi Rishi,

Even from commandline it throws the same error:
```
D:\github\serial-communication-manager\prebuilt-release>java -classpath sp-core.jar;sp-tty.jar;. test23.Test23
com.serialpundit.core.SerialComException: The parameter is incorrect.

        at com.serialpundit.serial.internal.SerialComPortJNIBridge.setRTS(Native Method)
        at com.serialpundit.serial.SerialComManager.setRTS(SerialComManager.java:1505)
        at test23.Test23.main(Test23.java:65)
```

RishiGupta12, 2017-08-02:
Any change in driver or hardware.

akuhtz, 2017-11-04:
Hi Rishi,
Will check on linux.

akuhtz, 2017-11-11:
Hi Rishi,
I tested on a raspi 3 and it worked without an exception.

RishiGupta12, 2017-11-12:
Found one more info https://stackoverflow.com/questions/2225737/error-jdwp-unable-to-get-jni-1-2-environment

Seems like it should have been fixed already by JVM/Windows/IDE etc. Should we close it.

akuhtz, 2017-11-12:
I see a similar exception with SPSW under Windows. Lets close it.

closed 2017-11-12 by akuhtz