RishiGupta12/SerialPundit #32: Get InputStream from SerialComManager

russellhoff, 2017-12-21:
I'm about to make use of marineapi. 

To start off, an InputStream is needed, but I don't know whether the first line is correct or not:
```
InputStream is = (InputStream) scm.getIOStreamInstance(SerialComManager.InputStream, handleGps, SMODE.NONBLOCKING);
SentenceReader sr = new SentenceReader(is);
sr.addSentenceListener(this);
sr.start();
```

If not, is there any other way?

Thanks!

RishiGupta12, 2017-12-21:
Example1: https://github.com/RishiGupta12/SerialPundit/tree/master/applications/byte-stream-binary-jar-integrate
Example2: https://github.com/RishiGupta12/SerialPundit/blob/master/tests/test53-byteStream/src/test53/Test53.java
Example3: https://github.com/RishiGupta12/SerialPundit/blob/master/tests/test54-closeport-Blockingread/src/test54/Test54.java

Expect some delay in my responses as I will be travelling to other country for vacations.

russellhoff, 2017-12-21:
Thank you for your help @RishiGupta12, I hope those examples are enough.

russellhoff, 2017-12-21:
I've taken a look at the examples, and `SerialComInByteStream` is always used. What I need is to get a `java.io.InputStream`. Is a simple cast enough (`InputStream is = (InputStream) scm.getIOStreamInstance(SerialComManager.InputStream, handleGps, SMODE.NONBLOCKING);`)?

RishiGupta12, 2018-01-01:
1. SerialComInByteStream extends InputStream in serial pundit. Any methods of InputStream are supported by SerialComInByteStream class (line 35 https://github.com/RishiGupta12/SerialPundit/blob/master/modules/serial/src/com/serialpundit/serial/SerialComInByteStream.java#L35)

closed 2018-01-04 by russellhoff