RishiGupta12/SerialPundit #33: "Can't find dependent libraries" on Win 10 32 and 64 bit

thkusch, 2017-12-24:
Hello!

A freshly installed Windows 10 (tested with 32 AND 64 bit) causes the following exception:

com.serialpundit.core.SerialComException: C:\Users\thomas\AppData\Local\Temp\sp_tuartx1\spcomwinx64.dll: Can't find dependent libraries
	at com.serialpundit.serial.internal.SerialComPortJNIBridge.loadNativeLibrary(SerialComPortJNIBridge.java:395)
	at com.serialpundit.serial.SerialComManager.<init>(SerialComManager.java:382)
	at serialTest.test.main(test.java:14)
Caused by: java.lang.UnsatisfiedLinkError: C:\Users\thomas\AppData\Local\Temp\sp_tuartx1\spcomwinx64.dll: Can't find dependent libraries
	at java.lang.ClassLoader$NativeLibrary.load(Native Method)
	at java.lang.ClassLoader.loadLibrary0(Unknown Source)
	at java.lang.ClassLoader.loadLibrary(Unknown Source)
	at java.lang.Runtime.load0(Unknown Source)
	at java.lang.System.load(Unknown Source)
	at com.serialpundit.serial.internal.SerialComPortJNIBridge$2.run(SerialComPortJNIBridge.java:378)
	at com.serialpundit.serial.internal.SerialComPortJNIBridge$2.run(SerialComPortJNIBridge.java:1)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.serialpundit.serial.internal.SerialComPortJNIBridge.loadNativeLibrary(SerialComPortJNIBridge.java:376)
	... 2 more

The source:
package serialTest;

import java.io.IOException;

import com.serialpundit.serial.SerialComManager;

public class test {

	public static void main(String[] args) {
		System.out.println(System.getProperty("os.name"));
		System.out.println(System.getProperty("os.version"));
		System.out.println(System.getProperty("java.class.path"));
		try {
			new SerialComManager();
		} catch (IOException e) {
			e.printStackTrace();
		}
	}
}

The output:

Windows 10
10.0
C:\Program Files\Java\jre1.8.0_151\lib\resources.jar;C:\Program Files\Java\jre1.8.0_151\lib\rt.jar;C:\Program Files\Java\jre1.8.0_151\lib\jsse.jar;C:\Program Files\Java\jre1.8.0_151\lib\jce.jar;C:\Program Files\Java\jre1.8.0_151\lib\charsets.jar;C:\Program Files\Java\jre1.8.0_151\lib\jfr.jar;C:\Program Files\Java\jre1.8.0_151\lib\ext\access-bridge-64.jar;C:\Program Files\Java\jre1.8.0_151\lib\ext\cldrdata.jar;C:\Program Files\Java\jre1.8.0_151\lib\ext\dnsns.jar;C:\Program Files\Java\jre1.8.0_151\lib\ext\jaccess.jar;C:\Program Files\Java\jre1.8.0_151\lib\ext\jfxrt.jar;C:\Program Files\Java\jre1.8.0_151\lib\ext\localedata.jar;C:\Program Files\Java\jre1.8.0_151\lib\ext\nashorn.jar;C:\Program Files\Java\jre1.8.0_151\lib\ext\sunec.jar;C:\Program Files\Java\jre1.8.0_151\lib\ext\sunjce_provider.jar;C:\Program Files\Java\jre1.8.0_151\lib\ext\sunmscapi.jar;C:\Program Files\Java\jre1.8.0_151\lib\ext\sunpkcs11.jar;C:\Program Files\Java\jre1.8.0_151\lib\ext\zipfs.jar;C:\Users\thomas\eclipse-workspace\SerialTest\bin;C:\Users\thomas\eclipse-workspace\SerialTest\libs\gettext-commons-0.9.jar;C:\Users\thomas\eclipse-workspace\SerialTest\libs\jssc.jar;C:\Users\thomas\eclipse-workspace\SerialTest\libs\xercesImpl.jar;C:\Users\thomas\eclipse-workspace\SerialTest\libs\sp-core.jar;C:\Users\thomas\eclipse-workspace\SerialTest\libs\sp-tty.jar
com.serialpundit.core.SerialComException: (....)

Any idea?
Best regards,
Thomas

thkusch, 2017-12-24:
It looks like it's a Java problem. When switching to jre1.8.0_144, it works. Back to jre1.8.0_151 causes the error

RishiGupta12, 2018-01-01:
hi thkusch, Happy new year. If this is solved we will close it.

thkusch, 2018-01-01:
hello Mr. Gupta! Also for you: a happy New Year.

This issue is NOT solved. The software doesn't work with 1.8.0 151. I assume that it also won't work with future versions. Sticking to 1.8.0 144 isn't a solution.

RishiGupta12, 2018-01-03:
Please verify that all Java related path (JDK and JRE) are correct in your system. This seems like mis-configuration of Java.

thkusch, 2018-01-03:
Thank you! I'm open for any suggestion. Is there anything particular that I should check?

Please keep in mind that I use a clean windows installation. To be more precise, I have snapshots of a Windows 10 (32-bit and 64-bit) out of the box in a virtual machine (Parallels). In one case I installed the 1.8.0 151 in the other case I installed the 1.8.0 144 just by running the usual installation process of Java. No further manipulation on the system. The result is described above.
The software runs fine on 1.8.0 151 on MacOS.

I would appreciate any tool or piece of code that could clarify the reason of the problem.

RishiGupta12, 2018-01-04:
I am sure that it is related to system configuration. You yourself has seen it works with 1.8.0 151 in macos. If SP has some problem than it would not work in any OS. We basically need to understand why JRE is not able to find libraries in 151 build.

Best try would be directly using windows 10 1.8.0 151 build (not virtual machine).

akuhtz, 2018-08-29:
Hello,
I came across the same issue today and solved it installing the "Visual C++ Redistributable für Visual Studio".

RishiGupta12, 2018-08-30:
Thanks Andreas for concrete update.

mariusheil, 2021-04-22:
Hello, we had the same issue and it was not exactly solved with the previous answer. Luckily I came across this page https://www.logfly.org/doku.php?id=en:faq:serialpundit that gives the real solution to the problem:

You need to install the 2013 version of the Visual C++ Redistributable für Visual Studio. This is very important. We installed more recent versions and the problem was not solved. Download the correct installer from here:

https://www.microsoft.com/en-US/download/details.aspx?id=40784