RishiGupta12/SerialPundit #34: tty2com -- cannot read meta info

slavkoja, 2018-03-17:
Hi,

i just compiled and installed `tty2com` driver. All seems OK, I am able to communicate with `picocom` and by python `serial` library and no errors in log. But i am not able to read meta info, as described in README:
```
LANG=C head -c 46 /proc/sp_vmpscrdk
head: error reading '/proc/sp_vmpscrdk': Invalid argument
```
The `cat` provides the same error message. The `sp_vmpscrdk` permissions:
```
ls -l /proc/sp_vmpscrdk
-rw-rw-rw- 1 root root 0 mar 17 15:23 /proc/sp_vmpscrdk
```
Kernel version:
```
uname -rsvo
Linux 4.14.12-2 #1 SMP Sun Jan 7 15:31:02 CET 2018 GNU/Linux
```

RishiGupta12, 2018-03-17:
Please try this:
head -c 52 /proc/sp_vmpscrdk

slavkoja, 2018-03-17:
Yes, this works:
```
head -c 52 /proc/sp_vmpscrdk
xxxxx#00000-00001#00002-00003#2#x-x#1-6#1-6#x#1#1
```

closed 2018-03-17 by RishiGupta12