RishiGupta12/SerialPundit #35: Linux kernel driver problem

slavkoja, 2018-03-19:
When i try to remove all devices (by command from README), my kernel stuck, in logs i found:
```
Mar 18 22:55:59 [44248.147927] BUG: unable to handle kernel paging request at 000000ff0000ffa8
Mar 18 22:55:59 [44248.147942] IP: _raw_spin_lock_irqsave+0x1e/0x40
Mar 18 22:55:59 [44248.147946] PGD 0 P4D 0 
Mar 18 22:55:59 [44248.147952] Oops: 0002 [#1] SMP NOPTI
Mar 18 22:55:59 [44248.147956] Modules linked in: tty2com(O) binfmt_misc reiserfs dm_mod br_netfilter bridge stp llc fuse tty0tty(O) tun loop pci_stub vboxpci(O) vboxnetadp(O) vboxnetflt(O) vboxdrv(O) it87 hwmon_vid joydev usblp cp210x usbserial hid_generic usbhid hid snd_hda_codec_hdmi btrfs zstd_decompress zstd_compress xxhash xor raid6_pq pktcdvd nvidia_drm(PO) nvidia_modeset(PO) evdev edac_mce_amd kvm_amd nvidia(PO) snd_hda_codec_realtek kvm irqbypass crct10dif_pclmul snd_hda_codec_generic crc32_pclmul crc32c_intel ohci_pci ghash_clmulni_intel pcbc snd_hda_intel drm_kms_helper aesni_intel efi_pstore aes_x86_64 snd_hda_codec crypto_simd glue_helper cryptd snd_hda_core drm snd_hwdep ipmi_devintf snd_pcm xhci_pci snd_timer ehci_pci pcspkr ohci_hcd ipmi_msghandler efivars xhci_hcd r8169 snd ehci_hcd fam15h_power k10temp
Mar 18 22:55:59 [44248.148034]  soundcore mii usbcore sp5100_tco usb_common i2c_piix4 shpchp button acpi_cpufreq efivarfs initio BusLogic arcmsr aic7xxx aic79xx scsi_transport_spi sr_mod cdrom sg sd_mod pdc_adma sata_mv ata_piix ahci libahci sata_qstor sata_vsc sata_uli sata_sis pata_sis sata_sx4 sata_nv sata_via sata_svw sata_sil24 sata_sil sata_promise libata scsi_mod ext4 fscrypto jbd2 mbcache crc16
Mar 18 22:55:59 [44248.148080] CPU: 2 PID: 3173 Comm: bash Tainted: P           O    4.14.12-2 #1
Mar 18 22:55:59 [44248.148083] Hardware name: Gigabyte Technology Co., Ltd. To be filled by O.E.M./970A-DS3P, BIOS FB 10/31/2014
Mar 18 22:55:59 [44248.148087] task: ffff94244ac2a080 task.stack: ffffac4f84c5c000
Mar 18 22:55:59 [44248.148092] RIP: 0010:_raw_spin_lock_irqsave+0x1e/0x40
Mar 18 22:55:59 [44248.148095] RSP: 0018:ffffac4f84c5fdf8 EFLAGS: 00010046
Mar 18 22:55:59 [44248.148099] RAX: 0000000000000000 RBX: 0000000000000206 RCX: 0000000000000000
Mar 18 22:55:59 [44248.148102] RDX: 0000000000000001 RSI: ffff94249d8a5848 RDI: 000000ff0000ffa8
Mar 18 22:55:59 [44248.148105] RBP: 000000ff0000ffa8 R08: 0000000000000000 R09: 000000000000003e
Mar 18 22:55:59 [44248.148108] R10: ffffac4f84c5fcd0 R11: 000000000000003f R12: 0000000000000002
Mar 18 22:55:59 [44248.148111] R13: ffff9423b855f700 R14: ffffffffc16356e0 R15: 000000000000003e
Mar 18 22:55:59 [44248.148115] FS:  00007fbfe889f740(0000) GS:ffff9424aec80000(0000) knlGS:0000000000000000
Mar 18 22:55:59 [44248.148119] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Mar 18 22:55:59 [44248.148122] CR2: 000000ff0000ffa8 CR3: 000000010ce4c000 CR4: 00000000000406e0
Mar 18 22:55:59 [44248.148125] Call Trace:
Mar 18 22:55:59 [44248.148140]  tty_port_tty_get+0x19/0x40
Mar 18 22:55:59 [44248.148150]  0xffffffffc1633c91
Mar 18 22:55:59 [44248.148159]  proc_reg_write+0x3d/0x60
Mar 18 22:55:59 [44248.148166]  vfs_write+0xad/0x1a0
Mar 18 22:55:59 [44248.148173]  SyS_write+0x52/0xc0
Mar 18 22:55:59 [44248.148181]  system_call_fast_compare_end+0xc/0x6f
Mar 18 22:55:59 [44248.148185] RIP: 0033:0x7fbfe8177eb8
Mar 18 22:55:59 [44248.148190] RSP: 002b:00007ffd10039f90 EFLAGS: 00000246
Mar 18 22:55:59 [44248.148193] Code: ff 66 90 c3 66 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 53 9c 58 0f 1f 44 00 00 48 89 c3 fa 66 0f 1f 44 00 00 31 c0 ba 01 00 00 00 <f0> 0f b1 17 85 c0 75 05 48 89 d8 5b c3 89 c6 e8 be 03 9a ff 66 
Mar 18 22:55:59 [44248.148282] RIP: _raw_spin_lock_irqsave+0x1e/0x40 RSP: ffffac4f84c5fdf8
Mar 18 22:55:59 [44248.148286] CR2: 000000ff0000ffa8
Mar 18 22:55:59 [44248.148295] ---[ end trace 883a70a9d4231776 ]---
Mar 18 22:55:59 [44269.152493] INFO: rcu_sched detected stalls on CPUs/tasks:
Mar 18 22:55:59 [44269.152510] 	2-...: (0 ticks this GP) idle=77a/140000000000000/0 softirq=2551346/2551346 fqs=0 
Mar 18 22:55:59 [44269.152514] 	(detected by 0, t=5252 jiffies, g=1896230, c=1896229, q=4195)
Mar 18 22:55:59 [44269.152523] Sending NMI from CPU 0 to CPUs 2:
Mar 18 22:55:59 [44269.152535] NMI backtrace for cpu 2
Mar 18 22:55:59 [44269.152541] CPU: 2 PID: 3173 Comm: bash Tainted: P      D    O    4.14.12-2 #1
Mar 18 22:55:59 [44269.152542] Hardware name: Gigabyte Technology Co., Ltd. To be filled by O.E.M./970A-DS3P, BIOS FB 10/31/2014
Mar 18 22:55:59 [44269.152544] task: ffff94244ac2a080 task.stack: ffffac4f84c5c000
Mar 18 22:55:59 [44269.152549] RIP: 0010:0xfffffffefa758469
Mar 18 22:55:59 [44269.152550] RSP: 0018:ffffac4f84c5f280 EFLAGS: 00000046
Mar 18 22:55:59 [44269.152553] RAX: fffffffefec10000 RBX: fffffffeffc5b318 RCX: 0000000000ffc5cc
Mar 18 22:55:59 [44269.152555] RDX: fffffffefec10000 RSI: fffffffefa6de2b5 RDI: fffffffeffc5b300
Mar 18 22:55:59 [44269.152556] RBP: 0000000000000001 R08: 0000000000000000 R09: 0000000000000004
Mar 18 22:55:59 [44269.152558] R10: 0000000000000004 R11: ffffac4f84c5f2f0 R12: 0000000000000000
Mar 18 22:55:59 [44269.152559] R13: 0000000000000307 R14: 0000000000000007 R15: 0000000000000000
Mar 18 22:55:59 [44269.152562] FS:  00007fbfe889f740(0000) GS:ffff9424aec80000(0000) knlGS:0000000000000000
Mar 18 22:55:59 [44269.152563] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Mar 18 22:55:59 [44269.152565] CR2: 000000ff0000ffa8 CR3: 000000022615e000 CR4: 00000000000406e0
Mar 18 22:55:59 [44269.152566] Call Trace:
Mar 18 22:55:59 [44269.152588]  ? efi_call+0x58/0x90
Mar 18 22:55:59 [44269.152595]  ? virt_efi_set_variable.part.7+0x5e/0x110
Mar 18 22:55:59 [44269.152599]  ? efi_query_variable_store+0x123/0x180
Mar 18 22:55:59 [44269.152603]  ? efivar_entry_set_safe+0x161/0x1d0
Mar 18 22:55:59 [44269.152606]  ? efivar_entry_set_safe+0x161/0x1d0
Mar 18 22:55:59 [44269.152612]  ? efi_pstore_write+0x10a/0x150 [efi_pstore]
Mar 18 22:55:59 [44269.152615]  ? efi_pstore_write+0x10a/0x150 [efi_pstore]
Mar 18 22:55:59 [44269.152622]  ? pstore_dump+0x169/0x330
Mar 18 22:55:59 [44269.152626]  ? vprintk_emit+0x337/0x3a0
Mar 18 22:55:59 [44269.152630]  ? kmsg_dump+0xb3/0xd0
Mar 18 22:55:59 [44269.152635]  ? oops_end+0x6e/0xc0
Mar 18 22:55:59 [44269.152638]  ? no_context+0x180/0x400
Mar 18 22:55:59 [44269.152642]  ? __do_page_fault+0x37b/0x4e0
Mar 18 22:55:59 [44269.152648]  ? page_fault+0x2c/0x60
Mar 18 22:55:59 [44269.152652]  ? _raw_spin_lock_irqsave+0x1e/0x40
Mar 18 22:55:59 [44269.152656]  ? tty_port_tty_get+0x19/0x40
Mar 18 22:55:59 [44269.152659]  ? 0xffffffffc1633c91
[... 181 lines cut ...]
Mar 18 22:57:04 [44333.376323]  ? efi_call+0x58/0x90
Mar 18 22:57:04 [44333.376332]  ? virt_efi_get_next_variable+0x6e/0x120
Mar 18 22:57:04 [44333.376336]  ? efivar_init+0x11c/0x3a0
Mar 18 22:57:04 [44333.376342]  ? sched_clock+0x5/0x10
Mar 18 22:57:04 [44333.376347]  ? efivar_release+0x20/0x20 [efivars]
Mar 18 22:57:04 [44333.376351]  ? put_prev_entity+0x1e/0xe0
Mar 18 22:57:04 [44333.376357]  ? efivar_update_sysfs_entries+0x24/0x60 [efivars]
Mar 18 22:57:04 [44333.376361]  ? efivar_update_sysfs_entries+0x24/0x60 [efivars]
Mar 18 22:57:04 [44333.376366]  ? process_one_work+0x185/0x380
Mar 18 22:57:04 [44333.376371]  ? worker_thread+0x2e/0x390
Mar 18 22:57:04 [44333.376375]  ? process_one_work+0x380/0x380
Mar 18 22:57:04 [44333.376379]  ? kthread+0x118/0x130
Mar 18 22:57:04 [44333.376382]  ? kthread_create_on_node+0x70/0x70
Mar 18 22:57:04 [44333.376387]  ? exit_to_usermode_loop+0x6e/0xc0
Mar 18 22:57:04 [44333.376392]  ? ret_from_fork+0x1f/0x30
Mar 18 22:57:04 [44333.376396] Code:  Bad RIP value.
Mar 18 22:57:04 [44333.376408] RIP: 0xfffffffefa753ef1 RSP: ffffac4f8a297b20
Mar 18 22:57:04 [44333.376411] CR2: fffffffeffc6e0f6
Mar 18 22:57:04 [44333.376415] ---[ end trace 883a70a9d4231777 ]---
```
Kernel used:
```
uname -rsvo
Linux 4.14.12-2 #1 SMP Sun Jan 7 15:31:02 CET 2018 GNU/Linux
```

RishiGupta12, 2018-03-25:
I will check and resolve. I am super loaded with other work at this time.

RishiGupta12, 2018-06-02:
Hi Slavkoja,

I tried a lot but not able to reproduce. Can you give details about how you were using this driver. And can you build your kernel with following debug option enabled so that we can debug issue CONFIG_DEBUG_SPINLOCK, CONFIG_DEBUG_MUTEXES.