RishiGupta12/SerialPundit #36: 2018-03-25 20:40:59 JNI call ThrowNew failed to throw exception !, FormatMessageA(), error code : 15100

makarna35, 2018-03-25:
Hi;

I'm try only this:
`class HotPlugEventWatcher implements ISerialComUSBHotPlugListener {

    @Override
    public void onUSBHotPlugEvent(int event, int usbvid, int usbpid, String serialNumber) {

        if(event == SerialComUSB.DEV_ADDED) {
            System.out.println("DUT added, running automated tests for this device !");

        }else if(event == SerialComUSB.DEV_REMOVED) {
            System.out.println("DUT removed, running tests, if any, to be run after device removal !");
            System.out.println("usbvid : " + usbvid);
        }else {
        }
        
    }
}`

If I'm unplug my usb Device from USB port netbeans show this message:

> 2018-03-25 20:40:59 JNI call ThrowNew failed to throw exception !, FormatMessageA(), error code : 15100

RishiGupta12, 2018-03-25:
I have seen this on a misconfigured Windows system. Did you formatted your
computer. Did you tried this with eclipse IDE.

makarna35, 2018-03-25:
Windows not have any problem. I'm use only NETBEANS

makarna35, 2018-03-25:
NOTE: I use FT series usb-com converter. I'm plug and unplug for test.

makarna35, 2018-03-25:
And now I get this message:

> JNI call CallVoidMethod() in usb_hotplug_event_handler() , failed !

makarna35, 2018-03-25:
JNI call CallVoidMethod() in usb_hotplug_event_handler() , failed ! error cause of me I'm fix it but 

> JNI call ThrowNew failed to throw exception !, FormatMessageA(), error code : 15100
problem still exist

makarna35, 2018-03-25:
I'm dedect problem but I not have a solition. If use this scenario you can see problem:
start read with scm.readString(handle); in while (read nonstop) and try UNPLUG USB device after this you can see "JNI call ThrowNew failed to throw exception !, FormatMessageA(), error code : 15100" message
You have any solition?

RishiGupta12, 2018-03-26:
May you check if same problem occurs with eclipse also.

makarna35, 2018-03-26:
I not have Eclipse