RishiGupta12/SerialPundit #37: My app not work some windows 10

makarna35, 2018-05-30:
Hi;
My app works one more windows 10. But I see not work one PC why?:

C:\>java -jar EWC.jar
Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: C:\Users\muhar\AppData\Local\Temp\sp_tuartx1\spusbwinx86.dll: Can't find dependent libraries
        at java.lang.ClassLoader$NativeLibrary.load(Native Method)
        at java.lang.ClassLoader.loadLibrary0(Unknown Source)
        at java.lang.ClassLoader.loadLibrary(Unknown Source)
        at java.lang.Runtime.load0(Unknown Source)
        at java.lang.System.load(Unknown Source)
        at com.serialpundit.usb.internal.SerialComUSBJNIBridge$2.run(SerialComUSBJNIBridge.java:340)
        at com.serialpundit.usb.internal.SerialComUSBJNIBridge$2.run(SerialComUSBJNIBridge.java:1)
        at java.security.AccessController.doPrivileged(Native Method)
        at com.serialpundit.usb.internal.SerialComUSBJNIBridge.loadNativeLibrary(SerialComUSBJNIBridge.java:338)
        at com.serialpundit.usb.SerialComUSB.<init>(SerialComUSB.java:136)
        at ewc.Main.<init>(Main.java:239)
        at ewc.EWC$1.run(EWC.java:42)
        at java.awt.event.InvocationEvent.dispatch(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)

RishiGupta12, 2018-05-31:
> at com.serialpundit.usb.SerialComUSB.(SerialComUSB.java:136)
> at ewc.Main.(Main.java:239)

Does both computer are 64 bit or 32 bit Windows ?

RishiGupta12, 2018-05-31:
spusbwinx86.dll is present in jar file, your computer does not have some libraries installed. Please run dependency walker http://www.dependencywalker.com/ and see what windows libraries you have not installed.

closed 2018-05-31 by RishiGupta12

makarna35, 2018-06-02:
Thanks.
Problem fixed. Problem cause of missing MSVCR120.dll file