RishiGupta12/SerialPundit #39: javax.comm Implementation

Bodyash, 2019-08-08:
Hi, I'm trying to wrap SerialPundit into Javax.comm API, cause javax.comm works only on linx86 or solaris

I need it to be able to connect to serial devices running on any OS and x86 or x64 Java, because a lot of legacy code used javax.comm API.

Currently i cannot test it, but i have a question, in legacy code i have something like:

```
serialPort.addEventListener(myListener);
serialPort.notifyOnDataAvailable(true);
```

and in MyListener ->

```
public void serialEvent(SerialPortEvent arg0) {
	if(arg0.getEventType() == SerialPortEvent.DATA_AVAILABLE) {
		while(serialPort.getInputStream().available() > 0 ) {
		    byte b = (byte)serialPort.getInputStream().read();
                //SomeCode
                }
               //SomeCode
       }
}
```

In my implementation of javax.comm.SerialPort i have:

```
public void addEventListener(SerialPortEventListener spel)  {
		
		//TODO Handle EVENTS SerialPortEventListener wrap to ISerialComDataListener or ISerialComEventListener
			scm.registerDataListener(handle, new ISerialComDataListener() {
				@Override
				public void onNewSerialDataAvailable(byte[] data) {
					// TODO Auto-generated method stub
				}
				
				@Override
				public void onDataListenerError(int errorNum) {
					// TODO Auto-generated method stub
				}
			});
	} 
```

And now i see some problem. Cause in javax.comm API code i'm recieving event that data is available and then reading bytes from input stream.

But in your code i see onNewSerialDataAvailable(byte[] data), so that means that data is already readed, it's in data variable, so serialPort.getInputStream().read() will not work? Or Im wrong?

RishiGupta12, 2019-08-09:
You are correct, data has been already read.

Bodyash, 2019-08-09:
> You are correct, data has been already read.
> […](#)

Hmmm, okay, but there is another way:

`	@Override
	public void addEventListener(SerialPortEventListener spel) throws TooManyListenersException {
		//TODO: HANDLE more events from lineEvent.getXXX();
		ISerialComEventListener listener = new ISerialComEventListener() {
			SerialPortEventListener javaxListener = spel;
			@Override
			public void onNewSerialEvent(SerialComLineEvent lineEvent) {
				if( lineEvent.getDSR() == 1) {
					javaxListener.serialEvent(new SerialPortEvent(thisPort, SerialPortEvent.DATA_AVAILABLE , false, true));
				}
			}
		};
		try {
			scm.registerLineEventListener(handle, listener);
		} catch (SerialComException e) {
			e.printStackTrace();
		}
	}
`
Listening to line events, "understand them" and send it like javax.comm events.
I Think it will work.
Or there is another solution - copying data to local input\output stream

Thx for reply!

RishiGupta12, 2019-08-09:
Data and line events are two different things. Keep them separate.

RishiGupta12, 2019-08-15:
If your question has been answered please close the ticket.

closed 2019-08-22 by Bodyash