RishiGupta12/SerialPundit #46: Drop "SerialComManager.BAUDRATE baudRate" parameter of SerialComManager.configureComPortData() function

powerbroker, 2021-09-03:
The "SerialComManager.BAUDRATE baudRate" parameter duplicates the "int custBaud" parameter. Nothing prevents from invoking 
SerialComManager.configureComPortData() with e.g. BAUDRATE.B115200.getValue() as "custBaud" value.

The first scenario of using configureComPortData() which comes into mind is:
```java
DATABITS db = DATABITS.DB_8;
STOPBITS sb = STOPBITS.SB_1;
PARITY p = PARITY.P_NONE;
int userTypedBaudRate;

// Some code affecting db, sb, p and userTypedBaudRate here

...configureComPortData(handle, db, sb, p, BAUDRATE.BCUSTOM, userTypedBaudRate);
```

RishiGupta12, 2021-09-04:
I am unable to understand what is the real problem.

powerbroker, 2021-09-09:
Hi,

Could you please review the attached sample of utilizing SerialPundit 
library?
Please focus on the static *private void startCommunication(String ... 
args)* method, there are explanation comments.

This is most likely not a problem but usability issue.

I'm implementing a GUI tool for serial IO through RS-485, running a 
CDBUS host on PC. There are some rooms in SerialPundit API require extra 
coding. The extra coding definitely can be avoided with little changes 
in API design.

powerbroker, 2021-09-09:
for some reason web ui fails to attach a file... please use 
https://disk.yandex.ru/d/T4tMFgahl01Qfw